F5OEO/tezuka_fw #155: I want the version that supports R2/T2 full MIMO

Ali89Mahdi, 2025-09-09:
Hello...

I would like to comment on a comment from the firmware owner, who says that you can currently only activate MIMO via an SD card.

I've tried most versions, and it still won't activate. Can you guide me through it, or provide me with the BOOT.bin/HDL that supports it?

My device is Pluto Plus.

Thank You

TeddySpaghetti11, 2025-09-21:
I'm not sure if this was a typo from Tezuka but I found that the full MIMO only worked when the firmware was flashed to the SDR. I am personally on the LibreSDR, but when running with an SD card, I couldn't get 2xMIMO working, but when I flashed it, I could run all the 2xMIMO examples on SDRangel (2 source DOA and beamforming). Hope this helped somewhat

F5OEO, 2025-09-21:
Confirmed https://github.com/F5OEO/tezuka_fw/issues/153#issuecomment-3316209332

F5OEO, 2025-10-25:
Closed to inactivity

closed 2025-10-25 by F5OEO