F5OEO/tezuka_fw #156: LibreSDR: incorect sampling of decimated stream in maia since 0.1.8

0x0V3RF10W, 2025-09-13:
Hello!

Normal visualisation and recording (checked in IQEngine) for SR 61.44 in 0.1.6.1, 0.1.8 and 0.1.9
[image]
[image]

Normal decimation for SR 30.72 in 0.1.6.1:
[image]

Abnormal decimation for SR 30.72 in 0.1.8 and 0.1.9
[image]

Please fix if it possible

F5OEO, 2025-09-21:
This is a bug which is difficult to solve. This is due to 2r2t management. Only happens on Libresdr.

F5OEO, 2025-10-25:
@0x0V3RF10W Can you check with 0.2.1 ?

0x0V3RF10W, 2025-10-25:
Hello, @F5OEO!
1. Thanks for the fix — works perfectly now. Much appreciated!

61.44 and 30.72 
[image]
[image]

2. One small note:
I’ve noticed another detail — the I/Q file size is now limited to 128 MB instead of 256 MB as it was before.
Is it ok for libresdr?
[image]

F5OEO, 2025-10-26:
Libresdr has only 512Mbytes RAM. Some linux distributions using much memory could not handle this amount of reserved memory. It could be set to 512 again, but for now easiest way is too lower it even on distribution like Tezuka which don't use memory. Let me know if it is a real drawback.