F5OEO/tezuka_fw #157: SDR++ Crashing when clicking play (LibreSDR)

TeddySpaghetti11, 2025-09-18:
Hi everyone, I am using a LibreSDR (ZYNQ7020, AD9363 configured as AD9361). 
When I boot up SDR++, and I press play after selecting the plutoSDR as my source, even at a sampling rate of 1MBPS, it plays for an instant and then crashes. Does anyone have any ideas on how to fix this?

76nic, 2025-09-18:
I have LibreSDR running on SDR++ without any problem. I just tested it again via USB and also booted up via flash and SD. All is working fine for me. I use SDR++ v1.2.1 on Kubuntu.
Best regards,
DC0NG

TeddySpaghetti11, 2025-09-18:
My apologies, I believe it may have been an issue with my pc power supply. I changed to an external power bank and it seems to work fine now.

closed 2025-09-18 by TeddySpaghetti11