F5OEO/tezuka_fw #160: 2 LibreSDR-branded units work with 0.1.5.2 (eMMC and sdcard) but later releases show an empty iio stream

twi42, 2025-09-28:
Hi,

I have two Hamgeek LibreSDR units, which have identical leds and connectors (including 10M/PPS input), and differ only in the aluminum casing.

Both work with earlier 0.1.5.2 release, apart from 10M reference input (which does work with with shipped, older Pluto release). However, installing any later Tezuka release initially works, but after restarting them a couple of times, I can connect with eg. SDR Console or SDRAngel, but only get an empty IIO stream. This happens both via Ethernet and USB. When the problem manifests, there are no unusual messages on the serial console and the only way to recover is to downgrade.

What might have changed since 0.1.5.2 to cause this?

Best regards,
Pasi OH2TWI

twi42, 2025-09-29:
Hi,

This happens without the network too, in usb rndis mode.  Powering LibreSDR off/on does not help once it gets into this state, only downgrading to 0.1.5.2 fixes it. However when accessing Maiasdr  only, the waterfall is also empty with later versions - but peculiarly, only after 1-2 restarts.

F5OEO, 2025-10-26:
Last versions use mode 2r2t by default. Even with SD card uEnv.txt doesnt overwrite env config. Thus, mode is not taking into account. Need to inspect and create a recovery mode to force setenv variables.
Setting standard mode 1r1t is solving the issue.

closed 2025-10-26 by F5OEO

twi42, 2025-10-26:
Weirdness, after exactly 2 reboots, I again get an empty waterfall with 0.2.1, even though mode is 1r1t. Changing it to 2r2t gives about one second of IIO stream, then nothing. 

Only way to recover is to downgrade to 0.1.6.

F5OEO, 2025-10-26:
Ok..need really to make the recovery mode. Stay tuned ! Sorry for inconvenience.